fbergama/pigfx #1: uspi @ 6620411 not found

foxweb, 2016-09-08:
I tried to make a build but uspi did not building.

[image]

It looks like a mismatch uspi submodule with your project.

(OS X El Capitan, cross-compiler toolchain was installed)

fbergama, 2016-09-09:
I confirm the problem. For a strange reason the uspi commit I was linking is no more present in their history.

I'll fix this asap. In the meanwhile you should be able to solve by doing the following:

```
git submodule update --init
cd uspi
git fetch
git pull
./make all
``` 

optionally, you may need to change `arm-linux-gnueabihf-` with `arm-none-aeabi-` in uspi/Rules.mk

foxweb, 2016-09-09:
Thanks!

Set `PREFIX ?= arm-none-eabi-` in `uspi/Rules.mk` and `./makeall` was effect. uspi builded successfully.

But make a build of pigfx was not so succefully:

[image]

fbergama, 2016-09-13:
Can you please list me the output of the following command?

`$  arm-none-eabi-gcc -print-libgcc-file-name`

it looks like the gnu c library is not linked properly

foxweb, 2016-09-13:
`/usr/local/gcc-arm-none-eabi-5_4-2016q2/bin/../lib/gcc/arm-none-eabi/5.4.1/libgcc.a`

fbergama, 2016-09-13:
Your error looks quite strange to me. I can't understand from your listing if it's at compile or link stage. But I can tell you that all those undefined references are from the lib gcc that should be provided by libgcc.a

I've tested my code both on Mac OS X El Capitan (your version!) and Linux Ubuntu xenial with no issues, so I'm wondering how to replicate the problem on my setup. 
Tomorrow I'll do some more experiments on that. In the meanwhile, in the bin directory you can find a pre-compiled binary ready to be copied on a raspi sd card.

foxweb, 2016-09-13:
Hmmm... compiled successfully with monkey patching:

`src/gfx.c`:
```c
inline void __swap__( int* a, int* b )
inline int __abs__( int a )
// replaced with
void __swap__( int* a, int* b )
int __abs__( int a )
```

`src/console.c`
```c
char buffer[32] = {0};
// replaced with
char buffer[32];
```

May be I have wrong versions of toolchain's utilities? For example:

```sh
$ ctags --help                                                                                                                                 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ctags: illegal option -- -
```

Sorry, I'm not so skilled in c-magic and toolchains. I'm use hombrew for installing tools for web-developing (lib-some-thing and etc). May be some conflicts with it?

foxweb, 2016-09-13:
[image]

TheCodeman, 2016-09-14:
If you use the same version as the author is builds fine .
https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q2-update
ken

fbergama, 2016-09-14:
Ok it depends on your toolchain. 

@TheCodeman: do you suggest to add the following definitions:
`
inline void __swap__( int* a, int* b )
inline int __abs__( int a )
// replaced with
void __swap__( int* a, int* b )
int __abs__( int a )
`
somewhere in the source code? I'm afraid that this may break the compatibility with some toolchains. For now, I don't want to use a configure script to figure our if such definitions are available or not.

fbergama, 2016-09-14:
I've patched the code in the [issue_1_fix](https://github.com/fbergama/pigfx/tree/issue_1_fix) branch. Can you please test it if it's working on your setup before I merge on the master branch?

Thank you

foxweb, 2016-09-14:
Ok, @fbergama your patch is fixed bug.

[image]

foxweb, 2016-09-14:
My version:

```
gcc version 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715] (GNU Tools for ARM Embedded Processors)
```

fbergama, 2016-09-14:
I've just merged the fix into master.

closed 2016-09-14 by fbergama