fbergama/pigfx #2: Pi Zero version 1.3

TheCodeman, 2016-09-14:
pigfx does not seems to start on the new pi zero v1.3 
only some random junk on the screen .
ken

fbergama, 2016-09-14:
I've tested it successfully on a plain old Pi first model and Pi Zero. I thought that the 1.3 version is no different from those except for the camera connector. 

It may be due to different addresses of some hardware peripherals. Can you do some tests?

foxweb, 2016-09-14:
I have 4 RPi boards (1, 2, 3, Zero) and I will try all of them (but I don't promise).

fbergama, 2016-09-14:
I can already tell you that it won't work on raspi 2 and 3 due to different peripheral addresses.
This can be fixed but I don't have time right now :(

foxweb, 2016-09-14:
So, I have Pi Zero v1.3 and SD-card with PiGFX from Spencer. This kit has been started properly at once and not modified firmware or changed Pi board.

[image]

fbergama, 2016-09-14:
So, you confirm that it works on the Pi 1.3 with the unmodified firmware you received by spencer. Right?

Can you please test the following:

- update your repo to the latest commit of pigfx/master 
- copy bin/kernel.bin to your raspi sd card. Leave all the other files untouched
- turn on the whole system and check if everything works

Thank you for your effort :)

foxweb, 2016-09-14:
> So, you confirm that it works on the Pi 1.3 with the unmodified firmware you received by spencer. Right?

Yep!

Please wait a few hours.

fbergama, 2016-09-14:
Ok so it's definitely not an hardware problem. If my kernel.img works it must be related to TheCodeman toolchain/system

TheCodeman, 2016-09-14:
Just reinstalled and rebuilt it all and seems ok now ? 
Not sure what it was . 
ken

fbergama, 2016-09-14:
Good news!
So.. I'll close the issue.

@foxweb: the test is not needed anymore, but thank you very much for your effort 👍

closed 2016-09-14 by fbergama

foxweb, 2016-09-14:
@fbergama @TheCodeman I have built the latest `kernel.img` and it works so good as unmodified firmware from Spencer.

[image]

fbergama, 2016-09-14:
Great! Thank you