fbergama/pigfx #3: Cannot print ESC literal. Missing up-left box character.

Herobrine2Nether, 2017-03-28:
I can print the up-right character with ^Z, and the X character with ^\ yet ^[ waits indefinitely for an escape sequence. Can a time-out or additional sequence be implemented to allow the up-left box character to be printed?
[image]
[image]

fbergama, 2017-03-30:
You are right, I assume that after the escape character an escape sequence occurs. Hence, there is no way now to output the escape char itself. 
I haven't got much time now, but it should be an easy change to fix

fbergama, 2017-03-31:
Ok if you update to the latest commit on master you should be able to output the ESC character by typing the ESC twice. Hope this closes the issue.

closed 2017-03-31 by fbergama