fbergama/pigfx #4: Serial port data not working

JmanNZ, 2017-04-08:
Hi
I am having an issue with a PI zero 1.3. If I connect the TX and RX the display echoes the keystrokes.
If I use any external device (ie FTDI or other serial device) the serial data is bad.
I connected a logic analyser to the Pi TX and to a FTDI and compared the outputs. Please see attached screen shots. I am using your prebuild version. Both system were outputting an M 

[image]

[image]

Regards
John

andresdelcampo, 2018-03-06:
I am not sure if it is the same problem, but my Pi Zero v1.3 was not outputting anything at all. It turns out the solution described by Brian Welsby in https://groups.google.com/forum/#!searchin/rc2014-z80/pi$20zero$20config%7Csort:date/rc2014-z80/0I-RAq3DnvU/9g2eLf1-AQAJ made it work! Worth a try...

Regards, 
Andrés

fbergama, 2018-03-07:
Sorry guys but I had limited time to work on this project lately. Thank for your contribution!  The readme file was updated as well

closed 2018-03-07 by fbergama