fbergama/pigfx #5: README doesn't list compatible Raspberry Pi models

MustBeArt, 2017-11-30:
There's nothing in the README (or any other document I found) to explain which versions of Raspberry Pi are compatible with PiGFX.

There's a mention of "all Raspberry Pi models" in the warning about serial port voltage, which seems to suggest that it runs on all of them. But I found an issue comment where you said it would not work on model 2 or 3, and I saw in an RC2014 document that it doesn't work on the Pi Zero W. Unfortunately I didn't find that issue comment until after I'd struggled for a while trying to get PiGFX to work on a Pi 3.

huskeyw, 2019-02-20:
I cant get it to work on any pi, just a rainbow screen at best

chregu82, 2020-04-30:
It is only working on a Raspberry Pi Generation 1 model at the moment.

chregu82, 2020-05-05:
It's written in the README now that at the moment only generation 1 is supported.

closed 2020-05-05 by chregu82

MustBeArt, 2020-05-05:
It works on Raspberry Pi Zero as well. (But not Pi Zero W). That’s what the RC2014 serial terminal board based on PiGFX uses.