fbergama/pigfx #7: Sending CR/LF when CR is pressed

CubeCentral, 2018-03-26:
I just built the Pi Zero Serial Terminal Module for my RC2014 and noticed that every time I press the "Enter" key to submit a command to CP/M, it is sending a CR followed by a LF.
This causes things like a directory listing to terminate after the first entry, among other artifacts.
Is there a way to configure this at boot time via the config.txt file?
I have put the most recent kernel.img from here onto the SD card I received as part of the RC2014 kit with no change.
I am using a Raspberry Pi Zero v1.3 and the RC2014 SIO2 card with the Pro backplane.  No custom work has been done and the entire kit is as-is from Spencer at RC2014.
I can work-around this issue by typing "CTRL-M" rather than "ENTER"

I also have noticed a strange keyboard mapping when using a standard (EN_US) USB keyboard where the double-quote character (") is mapped to SHIFT-2 (@) and vise versa.  I can work-around this issue as well, but it is a strange keystroke compared to what I assumed is a standard keyboard layout.

If I understood more about how to compile/make from source I could investigate further, but I am not that well versed in this type of software development.  I apologize if this is not the appropriate place to report this issue.   Thank you in advance for your kind attention to this matter.

CubeCentral, 2018-03-26:
I was informed that a fix to this issue could be found here: 
> https://groups.google.com/d/msgid/rc2014-z80/0e89ee77-6c4d-4caa-9b22-3f3334bb6346%40googlegroups.com

However, the double-quote issue persists with that version as well, but is easily worked around.

lurch, 2018-06-02:
To solve the 'wrong' character being sent, it looks like you'll need to change which keymap is being used https://github.com/rsta2/uspi/blob/master/include/uspios.h#L43
(I've not done this myself - I just stumbled across this project this afternoon and thought it looked interesting!)

chregu82, 2020-06-04:
The CR/LF behaviour and the keyboard layout both is now configurable in pigfx.txt.

closed 2020-06-04 by chregu82