fbergama/pigfx #8: Character Glitches?

wawa1474, 2018-07-06:
Unable to replicate the screen that is shown on the main page, certain characters don't show or weirdly reset the display.

wawa1474, 2018-07-22:
Moved this to my fork as it seems more appropriate. https://github.com/wawa1474/pigfx/issues/1
Plus this place seems dead...

closed 2018-07-22 by wawa1474

fbergama, 2018-07-23:
Hi,
The screen in the main page is just a sample of how the characters could be rendered on the screen. The terminal ansi codes parser do not allow you to print all the available characters. The obvious example is the ESC (27) used to start a sequence, or the newline character that would change the cursor position.

At the moment there is no way to just output the raw UART data. If you want to modify the sources, I suggest you to call the function

void gfx_putc( unsigned int row, unsigned int col, unsigned char c )

to manually render a specific char at a certain screen position.

I don't have so much spare time to support this project anymore, so any fork is welcome