fbergama/pigfx #13: Uart seems to be set for 7 bit + 1 stop

willtco, 2019-02-17:
After setting up the system I found the Pi was printing occasional extra characters, I hooked up my scope whilst connecting pins 8&10 (to form a 'loopback' test and the timings clearly show it is only set for 7 bits. The source code (uart.s) appears to be set correctly for 8 bit data.  At 115200 Bd one expects the total character time to be  1/(115200*10)   i.e.  1 start, 1 stop + 8 data to equal 86.8uS,   however the actual character time is (as in the picture approx 78uS and thus equal to only 9 bits. The clock frequency was double checked and was set at 3000000.  
[image]

willtco, 2019-02-18:
Apologies for posting in error.

fordp2002, 2019-02-18:
When set to 8 bits you will only ever see 9 bits on the scope. The stop bits is the same level as the idle bit so you can only ever see the start of the stop bit. If you send consecutive bytes you can sort of see the end of the start bit as it is the same as the start of the start bit for the next character.

willtco, 2019-02-20:
Yes,  I realised that (eventually), and tried to delete the post but, it does not seem possible - thanks for the reply.

chregu82, 2020-05-09:
This doesn't seem to need attention anymore.

closed 2020-05-09 by chregu82