fbergama/pigfx #14: this does not work

huskeyw, 2019-02-21:
raspberry pi zero 1.3
get nothing but rainbow.. have tried it with every file I can think of, installed on each raspberry pi I have to, I can get noob and any other OS to run.. but this does not work

CGizmos, 2019-02-22:
It works for me. Have you tried different SD cards? Do you have all of the files (not just the one) in the root on the card?

huskeyw, 2019-02-22:
I have gone through 4 cards it boots linux on these cards just fine.. could
my keyboard be an issue with boot?

CGizmos, 2019-02-22:
Mine boots without the keyboard. 

The files on the card:
bootcode.bin
config.txt
kernel.img
start.elf

I happen to be using an early model Pi because I didn't have any use for this particular board.

huskeyw, 2019-02-22:
Could not get it to boot on a 1st gen po, 2nd, 3rd or pie zero 1.3

huskeyw, 2019-02-22:
ok I got it working, incase someone else has the same problem.. 
so while the microSD works for linux.. this did not like it.. and not it.. in particular.. I removed all partitions and made a 1gig partition only.. formatted as fat (not fat 32 but fat) and now it boots and works

https://www.dropbox.com/s/kuufatufvrmozcc/20190222_132451.mp4?dl=0
https://www.dropbox.com/s/lxpfhm6gyavaxz8/20190222_132836.mp4?dl=0
 https://www.dropbox.com/s/6xnb34w8r8ytfi3/20190222_134045.mp4?dl=0

bkg2018, 2019-03-05:
Sorry to be late on this issue I'm quite busy on this year's beginning. 
The PiZero will boot with or without a USB keyboard. If you want to use one you must boot with the keyboard already attached: it won't hot plug. I use my PiZero board without keyboard, via a 
 serial/FTDI cable on the Serial board and on the PiZero board, with the RX jumper on the right (FTDI) side and the TX jumper on the lefft (PI) side, so I can receive datas from my terminal program on my Mac. 
About the microSD, I just checked and I use a 32GB formatted as FAT16 with one 1GB partition just like you do, although I don't really remember how/when/why I did this. I'll update the README for this.

huskeyw, 2019-03-09:
so new issue, this happened while testing colors and lines..  I have three brand new pi Zeros.. and while running the system rebooted.. and I get this https://www.dropbox.com/s/8grdggr0ggk10a3/20190308_213709.mp4?dl=0

just a Charater on the screen.. there is nothing connected but the pi.. I recreated from new everything.. new pi, new card, new files downloaded, new partition.. the PI boots NOOB zero issues..

GrantMeStrength, 2019-04-12:
I couldn't get it to work at all. Would just hang. Then I tried formatting the SD-CARD on a PC, not a Mac, and it started working.

huskeyw, 2019-04-12:
Got it to work by making small boot partition, then no issues, I dont have
a mac, linux and windows is all I have

rolfmobile99, 2019-12-02:
Hey Filippo and Francis,
I love the idea of using a Pi as a simple terminal.
But, I can't get it working.
I have a Rasp Pi 3, not a Pi Zero.
I have formatted a 16GB SD card with FAT32, and put the 4 files on it.
That didn't work (just saw a rainbow colored screen).
Then I tried again by reducing the partition to about 1GB (hard to get the exact numbers in fdisk).
I copied the same files onto the card again. Didn't work.
I'm really not sure what to try next?
Does this only work with Pi Zero?

I should mention that I got bootcode.bin and start.elf from
 https://github.com/raspberrypi/firmware.
(they weren't in your "bin" directory as your Readme says)
I have no idea if those are compatible with your kernel.

-Rolf

huskeyw, 2019-12-02:
I had the same issue, turned out that it's very sensitive to the type of
sdcard. I created a boot partition. And a main partition. The boot
partition I think is 512k..

bkg2018, 2019-12-03:
I just checked on a SD card from the PiZero Terminal I got recently from Spencer : it’s a 1GB card formatted in FAT (not FAT32). However I use 32GB cards formatted FAT32 with no particular problems on the PiZero 1.3.

Still, PiGfx doesn’t start on my PI3B neither. I’m no specialist of ARM assembler language but think the problem is probably that PiGfx is really close to the PiZero CPU. 

You can use Raspbian on your Pi, and an FTDI USB adapter to get a serial connection to the RC2014 : the USB goes in your Pi, and the FTDI goes to the SIO ftdi connector on the RC2014. It should create a /dev/tty*something* in your Raspbian which you can link to in your terminal software. I never tried that, but I do something similar on my macbook. All Unix-like OSes benefit from their 1970 ancestor  teletypes-oriented design and are natively good with serial devices.

Unfortunately, I have no idea of how to make PiGfx work on a PI3. That is too far from my knowledge level.

rolfmobile99, 2019-12-04:
Merci Francis!
I didn't realize that PiGFX was specific to the Pi Zero.   Ahhh ok.
So, I ordered a Pi Zero and hope to get it by end of the week!
I appreciate the quick feedback.
By the way, I met Spencer at a recent conference in Pasadena.. nice guy and cool Z80 project.
-Rolf

PS:  huskeyw - thanks for your answer too. It's nice you got yours working.

fbergama, 2019-12-04:
Hi all,
PiGFX was developed for raspi model 1 and Zero. It can be ported to other platforms (2 and 3) but this requires some changes in the initialization code. Unfortunately, I've no time now to take care of it, so the easiest solution is to just buy a Pi Zero :(

Good luck

chregu82, 2020-05-24:
PiGFX now works on all Raspberry Pis generation 1 to 3.

closed 2020-05-24 by chregu82