fbergama/pigfx #15: using h code

huskeyw, 2019-02-24:
so I am trying to figure out how to use the codes.. maybe I am in the wrong mode..  but I put in

print chr$(27)+"[?#x0;y0;x1;y1l"

and I get 
x0;y0;x1;y1l   as an output

bkg2018, 2019-02-25:
Hello, the sequence for graphic line is esc[#, not esc[?#.

You will find the list of supported code [here](https://github.com/fbergama/pigfx/blob/master/doc/terminal_codes.txt)

huskeyw, 2019-02-25:
Thank you for the responses I see the updated codes... but I dont get a
line ..

bkg2018, 2019-03-03:
I checked and found that STR$() inserts a space before the number too, while print inserts a space beffore and after any numeric value. So to send pure numbers with no space I had to use MID$(STR$(number),2). This is rather annoying but any space will end the escape sequence. Maybe I'll fix that so it's easier to send sequences, but it could have side efffects.

I did a little program to display lines of colors, in 640x480. This will show you how to send sequences to draw lines. Because of some line length limitation I had to break the drawline sequence in two lines : this is easy if you don't forget to end PRINT with a semi-colon ';'. I also end with a 70GOTO70 to avoid breaking the graphics at the end of the program. Just hit Ctrl+C to break away from the endless loop when you're tired of looking at your screen.

9 REM switch to 640x480
10 E$=CHR$(27):PRINT E$;"[=2H";
20 X=0
30 FOR Y=0 TO 479
39 REM set color: <ESC>[38;5;<n>m
40 PRINT E$;"[38;5;";MID$(STR$(Y AND 255),2);"m";
49 REM draw line: <ESC>[#<x0>;<y0>;<x1>;<y1>l
50 PRINT E$;"[#";MID$(STR$(X),2);";";MID$(STR$(Y),2);";";
51 PRINT MID$(STR$(639-X),2);";";MID$(STR$(Y),2);"l";
60 NEXT Y
70 GOTO 70

huskeyw, 2019-03-04:
Sorry I did not see this response, just finished my build and will give a try.. also running cpm3 and BBC basic right now, lots of flexibility with BBC basic including in line assembaly.. so hoping to get some software out to test show this solution off..

huskeyw, 2019-03-07:
so I ran the code and this is what I get
https://www.dropbox.com/s/hpu5lnojgjyxe7v/20190306_204558.mp4?dl=0

bkg2018, 2019-03-07:
Nice video, it helps to see what happens.

I see the Pi starts fine with the 8x16 font (30 lines of text in 640x480). The difference is you run the test program in CP/M MBASIC, while I was using the ROM basic directly from SCMonitor.

The change of mode doesn't even clear the screen, so obviously something is different in the way both BASICs send their PRINT to the terminal, or maybe it's in CP/M serial driver. My first immediate guess would be that MBASIC or CP/M interprets or intercepts the ESC code or the ESC sequences it doesn't handle. But I have to check that. Most printers of the CP/M era used various control characters so it would have seriously handicaped all softwares.

I will check this as soon as possible. 

At worst we could use a useless-as-possible code like CHR$(255) or whatever, which probably neither CP/M nor BASIC would take the risk to intercept. Not sure it would be the greatest idea though, VT100 and related terminals also use characters in the 128+ codes for technical drawing, foreign code pages or whatever.

huskeyw, 2019-03-07:
let me try it on basic 4.7 not in CPM, I just used CPM so I could save it
locally,  I'll also look to see if there is a known issues with escape
codes and cpm 3.. I could go back to cpm 2.2

William "Steve" Huskey
Managing Member Dreadnought Brewing LLC

huskeyw, 2019-03-07:
Basic 4.7b loaded from eeprom source and it works https://www.dropbox.com/s/a2xvuei1tv7d37o/20190307_074931.mp4?dl=0

huskeyw, 2019-03-11:
9 REM switch to 640x480
10 E$=CHR$(27):PRINTE$;"[=2H";
20 X = INT(RND(1)*639)+1 : Y = INT(RND(1)*478)+1
30 CO = INT(RND(1)*254)+1
35 X1 = INT(RND(1)*639)+1 : Y1 = INT(RND(1)*478)+1
39 REM set color: <ESC>[38;5;<n>m
40 PRINT E$;"[38;5;";MID$(STR$(CO),2);"m";
49 REM draw line: <ESC>[#<x0>;<y0>;<x1>;<y1>l
50 PRINT E$;"[#";MID$(STR$(X),2);";";MID$(STR$(Y),2);";";
51 PRINT MID$(STR$(X1),2);";";MID$(STR$(Y1),2);"l";
60 GOTO 20
this creates random lines and color.. maybe not the best ways as I guessed on color and lines numbers.. but it works