fbergama/pigfx #17: keyboard support change locals

huskeyw, 2019-03-12:
I can't find the def to change locals for my own build

chregu82, 2020-05-09:
Not quite sure what you mean. Do you want to change the keyboard layout?

huskeyw, 2020-05-09:
It's been a while but I think local was uk which means on my US board I had
to put a sticker for which key I used

chregu82, 2020-05-09:
The USPI library is compiled for UK keyboard layout. This is an external component. It could be changed to US by changing a define in a header file. But you would have to build your own binaries afterwards. There is no way to change this from inside PiGFX for now.

huskeyw, 2020-05-09:
So do a pull to my own. Your fixing a lot of stuff now so I'll wait ..
thank you

chregu82, 2020-06-03:
The keyboard layout is now configurable since version 1.4.0.

closed 2020-06-03 by chregu82

huskeyw, 2020-06-03:
I want to say your doing an awesome job, I use this for more than RC2014
and used it to create a card for the z80-mbc2 as well
William "Steve" Huskey
Managing Member Dreadnought Brewing LLC