fbergama/pigfx #20: 1024x768?

kolontsov, 2019-11-23:
Hi, thank you for pigfx, it works well, but it doesn't look very good when scaled.. Is implementing 1024x768 difficult because it's bare metal? What is the main obstacle?

chregu82, 2020-05-09:
I just gave this a test. It would be easy to include 1024x768 mode. Graphics are quite slow though and the text is quite small.

willtco, 2021-09-13:
Can you briefly list the steps required to achieve 800x600 ?

chregu82, 2021-09-13:
Without having tested anything, I would try the follwing:
- Add the resolution in gfx.c to the enum DISPLAY_MODE_DEFINITION, e.g. replace element 7 with {800,600,8}.
- Send esc[=7h to the Pi

willtco, 2021-09-13:
Thanks,  I had found  initialize_framebuffer in pigfx and altered that to 800x600 which seems to work fine - is that reasonable?

chregu82, 2021-09-14:
That's probably fine.

willtco, 2021-09-14:
I've added a simple 3 byte command that allows me to select any 1 of 16 resolutions from 320x240 to 1920x1080 - I use the card purely for graphics so all the escape sequences have been replaced with short binary packets. -  My screen res routine is below, once tested fully (along with now using  a PS/2 kbd)  I hope to offer it up for others. 

/*Function to allow Z80 to change screen mode by sending 3 bytes only. Added Sept 14 2021.
 Command string:  0xff,0xe3,0x40+mode  (where mode is 0-f only). */

int state_set_screen_resolution( char ch, scn_state *state ) //function E3
{
static unsigned char mode;
   lineparms[parm]=ch & 0x0f;  //;low 4 bits contain mode data
   mode=lineparms[0];
		
	switch (mode)
	{
	case 0x01:
		initialize_framebuffer(400,300,8);
		break;
	case 0x02:
		initialize_framebuffer(320,240,8);
		break;
	case 0x03:
		initialize_framebuffer(640,480,8);
		break;
	case 0x04:
		initialize_framebuffer(800,600,8);
		break;
	case 0x05:
		initialize_framebuffer(1024,768,8);
		break;
	case 0x06:
		initialize_framebuffer(1280,800,8);
		break;
	case 0x07:
		initialize_framebuffer(1600,900,8);
		break;
	case 0x08:
		initialize_framebuffer(1920,1080,8);
		break;
	}

   state->cmd_params_size = 0; 
   state->next = state_fun_normaltext;

 return 1;
}