fbergama/pigfx #21: Installation issue, no HDMI output

HaViGit, 2020-01-09:
Need some help to install PIGFX on a RPi Zero 1.3 because I did not succeed and there is not any video output. These are my steps so far:

– using a 16GB fast Sandisk SD card I made only 1 partion (of 1 gb)
– FAT32 formatted
– copied bootcode.bin, kernel.img and start.elf in root of the SD card
– created a config.txt file containing one line: init_uart_clock=3000000
– copied config.txt to the root of the SD card

The Zero is powered with 5.1 volt / 2500 mA and after (re-)booting the Zero LED goes on for only a short time and that is it. Not any video output, with or without USB keyboard connected.

The bootcode.bin and start.elf files were extracted from another (working) Pi-Zero SD card installation. The kernel.img file was downloaded from https://github.com/fbergama/pigfx 

Someone knows if I am missing something?

bkg2018, 2020-01-10:
Hello,

Looks like it should work fine. Maybe I'd use a 5V power supply rather than a 5.1V. Ttry with a basic USB->micro USB cable (like the ones for samsung phones pre usb-c) on an USB port from any computer. 

Can you check if your PiZEro works ok with a Raspbian (on another SD card)?

HaViGit, 2020-01-10:
Thanks for replying Francis. The Pi Zero 1.3 is okay (I've tested two), the power is okay and powered via PC or powerplugs did not solve. 

Below some text about the issue as posted by me on a retro forum.

Regards, Hans.

_I have allocated a 0.9 GB partition on a Kingston SD card and on several fast Sandisk cards,  FAT and FAT32 format as well. The files bootcode.bin, start.elf, kernel.img and config.txt were copied to the root (/) of the prepared SD card. After rebooting the Zero the LED turns on for a short time, sometimes it keeps blinking but there was never any HDMI output. No matter how I setup the HDMI monitor.

To me it seems there are circulating several versions of the four needed files, I have tried many without succes including files from https://github.com/fbergama/pigfx._

chregu82, 2020-05-24:
If possible, please try this again with the latest version 1.2.0 of PiGFX.

chregu82, 2020-07-19:
V1.6.2 is working on Pi Zero 1.3.

closed 2020-07-19 by chregu82