fbergama/pigfx #22: PiGFX keyboard TX issues

zaphodbe, 2020-02-16:
If I load linux on the PI zero the serial port works fine and I can talk to my RC2014, however if I load PiGFX it receives the data from the RC2014 but seems unable to send anything to it.  

I tried the normal trick of shorting the TX and RX lines on the Pi to create a loopback and get the same results, works with Linux and not with PiGFX.

Tried several Pi Zeros (all non W) and all have the same symptoms.

Any ideas?

zaphodbe, 2020-02-18:
Ok found the issue.  Basically the issue is that for some calls to attach_timer_handler in the stupid_timer.c file.  The hz parameter is 0 which results in an illegal divide in the following line and a very long time before the handler is called hence breaking the USB keyboard.

            timers[hnd].microsec_interval = 1000000 / hz;

So changed the code as follows to make sure the handler still get called in a timely fashion.  The 0 might need to be more that that but either way this makes sure the handler gets called next time around the poll.

            if (hz) {
                timers[hnd].microsec_interval = 1000000 / hz;
            } else {
                timers[hnd].microsec_interval = 0;
            }

chregu82, 2020-05-09:
This is fixes with Version 1.1.5.

closed 2020-05-09 by chregu82