fbergama/pigfx #25: Problem building

clausmunk, 2020-05-15:
I'm trying to build the latest version but get the following error:

cc1: error: '-mfloat-abi=hard': selected processor lacks an FPU

I'm using "arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]" under MacOS.

Any hints?

chregu82, 2020-05-15:
Are you trying to build the master or the raspi2 branch?

clausmunk, 2020-05-15:
I have tried both. Same result. I also get the error when trying to build the uspi subcomponent.

chregu82, 2020-05-15:
Raspi2 branch is work in progress. It won't boot. I checked my gcc version, I'm on 7.4.0, which seems to work. You probably need to change some compiler options for newer gcc, but I'm not sure.

clausmunk, 2020-05-15:
Ok, I’ll try to dig up an old version of the tool chain.

Thanks

closed 2020-05-15 by clausmunk

chregu82, 2020-05-28:
PiGFX should now be buildable with the latest toolchain.