fbergama/pigfx #29: PiZero on newest bin file

lindoran, 2020-07-12:
I have a Pi Zero v1.3 I used in place of a model B.  The model b works fine, no issues PS2 and USB.  I put in the pi zero, and it would get locked up in boot.  If it would boot (witch was not often) it would have issues with key bounce on the USB.   I also tested the pi zero with Linux and it boots to desktop no issues and the keyboard works fine with the OTG adapter.  I tried multiple SD cards and even formated with GParted to make certain that the SD card was in a format that the Pi would accept.  I set the partition size to really low (like 8MB) and also tried 1GB and 500MB.  still no love would commonly crash after displaying the volume name, or just before. again no issues at all on the newer PI.

chregu82, 2020-07-13:
Hello. I never tested with a V1.3 Pi Zero. I try getting one.

chregu82, 2020-07-16:
This should be fixes with V1.6.1, please try this release.

lindoran, 2020-07-16:
The pi zero boots just fine, so that issue is fixed but I'm still running into issues with key bounce on the USB keyboard.  even momentary press of a key produces 3 to 4 key presses from a keyboard that works just fine on the model 3b.  thanks so much for your help

chregu82, 2020-07-17:
Thank you for reporting this bug. I'm quite sure, that this is solved with V1.6.2. Please try this version.
The startup screen gets cleared if at least 8 characters are received, maybe this is earlier than your screen is ready to show something with the pi zero. You could try booting the zero without connecting the serial port.

lindoran, 2020-07-17:
all is working fine with USB. Thanks For all your help!

closed 2020-07-17 by lindoran