fbergama/pigfx #32: Pi 3 Model B

lindoran, 2020-08-06:
on 1.7.0 the terminal will not load past showing the baud rate. on the Pi 3 Model B

chregu82, 2020-08-06:
I just tried this again and found no problems. Did you send at least 8 characters to the terminal? Baudrate is right? Is your ACT LED still flashing?

lindoran, 2020-08-06:
yes it is working on the pi zero but not on the Pi 3 Model B v1.2

lindoran, 2020-08-06:
i will try to make a video later today

chregu82, 2020-08-06:
My Pi3 Model B V1.2 works fine on V1.7.0. Is your LED still flashing? Is it working on V1.6.x?

lindoran, 2020-08-06:
I will try and get back to you :) probably a couple of hours

lindoran, 2020-08-07:
I went back and tried with 1.6.2 and the terminal is not working there so it must be a problem with the UART pins on my pi 3.  My apologies the good news is my pi zero does work so ill do the remainder of the testing with that thanks for checking on your end ill try to be more thorough in the future

closed 2020-08-07 by lindoran