fbergama/pigfx #34: sprites

lindoran, 2020-08-15:
what is the preferred method of loading sprites to the terminal? I am trying some of the sample files and it is not working correctly.  I tried just dumping them to the screen buffer using the type [NOPAGE] command from CPM.  which works well for the animate dinos.bin, but when you try to load the load dinos file several of the characters flood to the screen instead of being read as pixels.  Is there a way you could explain how to load a simple bit map from basic, C or Pascal, or explain a better way to get the sprites to load? I will try to make a video to show whats happening.

lindoran, 2020-08-15:
https://drive.google.com/file/d/1Xg6LNHMTUqN-xAakSZLLslO4duYN43rG/view?usp=sharing

chregu82, 2020-08-15:
Well, actually I never used a Z80 machine of any kind or even CP/M. I do all my tests with a terminal software on Windows. There may be problems, if the data is not sent binary. Commands like type probably don't just send the binary file. 
I found this about the type command: The TYPE command expands tabs, CTRL-I characters, assuming tab positions are set at every eighth column.
So I assume, this command manipulates the data if there's a TAB byte.
I also made the experience that the DOS type command doesn't work correct either, as it interprets one specific byte as end of transfer and stops.
So, all you have to do is find your method of choice for sending binary data to the terminal, unfortunately I don't know how to do this in CP/M.

chregu82, 2020-08-15:
After some googling I think maybe the PIP command could work. The main problem is, it must sent unmanipulated binary data.

lindoran, 2020-08-15:
I tried useing PIP to simply redirect the file to the device file for the console, PIP kind of frezes up; you can break with ctrl C, the sprite doesn't load however when you send the animate seqence nothing shows ... so it's sort of working .... but not as intended.  This issue is almost definitely CPM.   There are a few options in pascal and C I can try to load the sprite maps to the terminal but im expecting similar results due to how CPM takes control of the console port; its generally protected space in middle and high level programing languages.  there may be a way to directly access the port directly similar to the way the ZMODEM protocol works (I have used this interface with CPM and a TTL serial link to move files to and from a windows work station) but since the SBC i am using has only 1 serial interface it's not practical for this application (the computer is hooked to the PiGFX interface).  Would it be possible to do a simple character based transfer? an example would be simply sending a list of color values in ASCII after the sprite definition.  so for example <ESC>[#<idx>;<x>;<y>;{list of color values in decimal separated by semicolon}a <--- denotes end of command sequence.  obviously this would be very cumbersome for something really large, but a 8x8 tile, would only be 48 values less if you could figure a way to use a syntax that includes something similar to RLE (that's a big ask, I understand). The memory management on the client side could be paged or could be done with segmenting so that the memory is freed after the sprites are loaded.  This type of thing would make it much simpler to implement on very old character based systems, which have limiting control over the console.

chregu82, 2020-08-15:
I think I see the problem... As there are probably more users with CP/M who have limited access to the serial interface, I'm going to implement a text based bitmap loading interface. The rest of the sprite handling doesn't need any changes.
Your suggestion looks promising. However it'll probably be something like <ESC>[#<idx>;<x>;<y>a followed by a semicolon separated color pixel list with x*y elements. This is a very slow and byte-wasting way of loading a bitmap, but better than nothing.
This could be done with RLE compression in a similar way.

lindoran, 2020-08-16:
This is excellent news! Thank you for your time :)

chregu82, 2020-08-16:
Please try V1.7.4.

lindoran, 2020-08-16:
That was really fast! I'll try tonight when i get home from work 😃 thank
you so much.

lindoran, 2020-08-17:
This is working quite well! No issues so far, loading the dinos at the fastest terminal speed takes about a minute or two which is understandable given what all is happening.  the way I'll handle this in my first program is using smaller tiles and loading in pieces.  I think this will work very well!

closed 2020-08-17 by lindoran