fbergama/pigfx #36: autorepeat dosen't function as expected.

lindoran, 2020-09-01:
when you hold down a key, there should be a slightly longer delay followed by a series of repeated key presses.  an example would be if you open a dos window in windows or an xterm when you hold down the L key it should wait a few milliseconds and then start rapidly repeating that key press.  The same should happen with the backspace key. this functionality works like this on the pi terminal in Linux but not under pigfx.  The same is true for another serial terminal emulator i tested (the grant serle video and keyboard interface).  I do not have a way of testing with PS2, and am only experiencing this with the USB keyboard. this would be on the PiZero 1.3  If you try to toggle the backspace key echo in the config file it does echo the backspace 1 time, and then it does not anymore(it sends 2 of the currently configured control codes for backspace).  If you connect to the host computer with a terminal emulator running on a PC and a TTL to USB adapter it works as expected.  This issue happens under CPM ,BASIC and forth interpreters on tests were performed on a Z80MBC2 SBC, and putty configured for 115200, with control-H for backspace, on a windows terminal emulator for the same computer (the Z80MBC2).

chregu82, 2020-09-01:
I'm really not sure, I understood your issue.
I assume you have a USB keyboard connected to the Pi Zero and the Pi Zero connected to your Z80MBC2. If you hold down a key on this keyboard, it does not repeat this key? Which version of PiGFX are you using?

lindoran, 2020-09-01:
I am using the newest version with sprite collision, I haven't noticed
because I'd been using a notepad app on windows to edit source so I
couldn't say if it was always an issue or not.

chregu82, 2020-09-01:
I'm quite confused about this issue. I tested V1.7.5 on a Pi Zero 1.3 connected to a windows machine with an USB to serial adapter. I ran putty there. If i press a key on the keyboard for longer than about halb a second, i get repeated characters on the terminal on the windows machine. Maybe I didn't understand what you were trying to do.
The config.txt option backspaceEcho does a backspace on the Pi display AND sends the backspace to the serial output if backspace is pressed on the Pi keyboard.

lindoran, 2020-09-01:
Only thing I can think then is, it's something with the computer and the
way it's handles input from the terminal. Are you using the keyboard that
is plugged to a otg adapter on the PiGFX terminal or the keyboard on the PC
that is running putty? Another way to look at this is... When I test it
using only putty as a terminal it works fine on the Z80-mbc2 but when I use
the PiGFX interface and a keyboard hooked through an OTG adapter and a HDMI
monitor it dose not. That's especially troubling to troubleshoot because
putty works but PiGFX doesn't. I'll try more tests tonight. Is there some
way PiGFX differs from putty as a terminal emulator?

chregu82, 2020-09-01:
I'm still not quite sure, if I understand your problem and your setup. Is it possible to do a short video showing your setup and showing what doesn't work?
You write that it work if you connect putty to your Z80, but it does not work on PiGFX as the terminal. Do you refer to key repeat or backspace echo?

lindoran, 2020-09-01:
I'll absolutely do a short video. 😁 It's going to be later today work till
late afternoon. I'll get back to you as soon as I can.

chregu82, 2020-09-01:
That's fine, I'll probably be sleeping by then. 😁

lindoran, 2020-09-02:
here's the video, first is the z80 computer hooked to the PiGFX terminal
using the USB keyboard.  2nd part of the video is the same z80 computer
hooked to putty through a usb to ttl link, using the same 4 wire connection
as with the PiGFX terminal.  I believe this shows that the issue is not the
z80 computer.

https://drive.google.com/file/d/1cM4YVeGRTsEM1PzfLB8cdHiLbLgkYs4e/view?usp=sharing

chregu82, 2020-09-02:
Thanks for the video. This feature works in my setup. Please try a different USB keyboard if possible, I think it probably acts different than mine.
You could also try connecting your windows pc to the Pi zero. Typing on the keyboard would then be seen in putty on the Windows machine. Please only try this if your USB to TTL adapter is 3.3V.

lindoran, 2020-09-02:
I tried a different usb keyboard to the same results. my usb ttl adapter
has has a 3v3 line but I need to solder headers to it so that is for
later.  I did try hooking the terminals together last night with the by
directional shifter to try to replicate your results but I couldn't get the
terminal to go past the bootup screen ... Putty sees the link but I must be
doing something wrong because there are no characters on either side when I
type. I might have the rx / tx lines wrong I thought I tried crossing over
the link and doing straight through. I'll try more later.

chregu82, 2020-09-02:
Please try the following:
Disconnect serial lines from the Pi and connect RX and TX pins on the pi. What you type on the USB keyboard immediately gets displayed on the Pi display. Please check if key repeat works like this.

lindoran, 2020-09-02:
Will check and let you know

lindoran, 2020-09-03:
hooked up as suggested, what is odd is it didn't work then either.  But then  i tried the keyboard for my main workstation just to try something else and it worked! so that's a good thing.  I had tried 2 other USB keyboards and found 1 with the same issue and another that just didn't work at all. so im stumped.  All of these keyboards seem to work just fine under windows.  I am confident I can find a keyboard that works; I just have to keep trying keyboards.

chregu82, 2020-09-03:
I should probaly try different keyboard too. The seem to act different. With my keyboard I get events for key pressed and key released. I'm repeating until key is released.

lindoran, 2020-09-03:
Thank you for the help on this. 😁 At least there is a solution