fbergama/pigfx #37: Potential Feature? Making some of the keyboard more customizable.

lindoran, 2020-09-03:
Since discovering that certain keyboards behave differently with the interface, how hard would it be to make the keyboard denounce settings available in the .ini file.  things like manual key repeat through software (for keyboards that do not support this behavior), de bounce delays, repeat behavior or general key reassignments for odd keyboards (not just the backspace)?  Just a thought i was having, it should not be considered in any way a high priority.

chregu82, 2020-09-03:
USB keyboards never do key repeat on their own. It's always done in software. PS/2 keyboard on the other hand do key repeat on their own. If I had a non repeating USB keyboard, I probably could make it working. Something must be different from keyboard to keyboard.

lindoran, 2020-09-03:
I have several I could send you one if you would like for testing I would
pay the shipping

lindoran, 2020-09-03:
I I might also be able to run USB debugging and read information from the
keyboard that's not working and compare it to the one that is would that be
helpful?

chregu82, 2020-09-03:
Let's try option 2 first. This is a test version which outputs the raw keypress data to the screen. I would expect that on key press a hex number >0 is displayed, and on key release 00. Modifier keys like control or alt should display 00 too.
[https://1drv.ms/u/s!Anft_sBKDg5arwr47cL0xuunUqld?e=TIUav4](url)
Image is for Pi1 / Zero only.

lindoran, 2020-09-03:
the link doesn't appear to be working

chregu82, 2020-09-03:
I'm sorry, please try again.

lindoran, 2020-09-03:
ok so here's what happens ... on all the keyboards that don't support autorepeat. they continually send key release 00 unless you press a key then they continually send the hex code for that key... Ill send a video in a few minutes i have to upload it
the keyboard that works is exactly as you described.

lindoran, 2020-09-03:
https://drive.google.com/file/d/1whNoy8TbhnPC_fQVS_o2-RFWYeuaPC26/view?usp=sharing

chregu82, 2020-09-03:
I don't seem to have access to this file.

lindoran, 2020-09-03:
Sorry try this

https://drive.google.com/file/d/1whNoy8TbhnPC_fQVS_o2-RFWYeuaPC26/view?usp=drivesdk

https://drive.google.com/file/d/1whNoy8TbhnPC_fQVS_o2-RFWYeuaPC26/view?usp=drivesdk

lindoran, 2020-09-03:
I also found that a 00 is sent when releasing a key holding a modifier down
and releasing that modifier. So if you tap a key 00 is sent much faster...

chregu82, 2020-09-04:
Please try if this works for all your keyboard: [img](https://1drv.ms/u/s!Anft_sBKDg5arwr47cL0xuunUqld?e=TFUZCN)

lindoran, 2020-09-04:
tested and is working :) actually a bit better than the other keyboard seams far more responsive
thank you so much! its probably fine to push to current.  I tried out the modifier keys as well and caps lock everything works.
you work so fast! I wish I had your stamina.  I got a bit more done on my sprite editor tonight, but its like 1:20 in the morning where I am, im headed to bed. thanks again, much appreciated!

closed 2020-09-04 by lindoran