fbergama/pigfx #38: missing control code for insert line

lindoran, 2020-09-04:
a lot of full screen editors in CPM use the "insert line current line moves down" ANSI control code. I was playing around with settings in turbo pascal and I noticed this was missing.  It will cause an issue that causes the full screen editor to not scroll correctly.  It's not a Huge deal because most editors give a redraw screen key sequence (it is usually ctrl-kk) and will display the current line for positioning in the lowest portion of the screen, when scrolling so you can move to the line you want and redraw the screen by pressing the key sequence.   The most commonly used escape sequence is ( esc [ 1 L )  where the numeral is how many lines to insert, as documented https://www.inwap.com/pdp10/ansicode.txt.  It is again by no means 100% important, but something to think about because some of the older terminal based games use it as well, though not nearly as much as esc [2J witch i've found clears the sprites currently, it may be a good idea to have a separate escape code for that (clearing the sprites), but its just something you have to be aware of and when doing on screen graphics. you need to pay attention to so you don't accidentally wipe all of your sprites.  I know its a big ask you have done so much already.  It was just something that I noticed.

chregu82, 2020-09-04:
I'll have a look at these missing escape codes.
esc [2J currently has to clear the sprites too, as it overwrites the whole screen content with background color. It would be quite painful to redraw every active sprite afterwards. What I could do is a command for only clearing active sprites. Would that be of help?

lindoran, 2020-09-04:
Clearing the screen of everything is is fine, it is supposed to do that but
clearing the stored sprites is excessive because you then have to re load
them to memory on the pi ( a large hex bitmap can take 3 minutes or more
even with RLE.) Which is why I use smaller tiles loaded as needed and
generally I use data that's stored in the host computers memory as loading
from the SD card is very time consuming in this way as it greatly
contributes to load time.

chregu82, 2020-09-04:
There's no stored data removed. Only the defined sprites are removed. You can simply redraw them with the bitmap index you loaded before.

lindoran, 2020-09-04:
Ok, understood, I was mistaken; apologies.  Then the only issue are the
more specific missing control codes.  I also found that highlight on /off
is missing but it is often missing from terminal emulation and with color
I'm not sure how it would be implemented correctly, so it's probably safe
to leave off and just focus on insert line as it is used far more. The only
thing I could think was swapping the background / foreground  color... But
even then on some terminals the expected output is dim/bright intensity....
Which wouldn't highlight either. Sometimes they are two different features
one for intensity, one for highlight. Sometimes they are swapped in
terminal emulators...

chregu82, 2020-09-05:
I implemented the 4 commands for delete/insert character/line in V1.8.0. Please test.

lindoran, 2020-09-06:
This works as intended, however I was thrown a bit on the set up in turbo pascal --- but it was completely my fault.   I was using esc [2K to clear a whole line instead of esc[1M for erase line (scroll up).   The documentation is clear I was not reading for detail.  Another thing that kind of threw me but I fixed it, was that the standard text mode by default is 80x30 and not 80x25... so I was having issues with the whole screen being filled, after I referenced the chart in README_ADD.md (again which was very clear), I had no issues determining the screen dimensions.  Thank you again for all your help! 👍   I'm going to try do do a small video on the latest updates to the terminal so I can showcase them over my holiday.  I should have a few moments to shoot and some time to edit on Monday hoping to have it up by Wednesday.

closed 2020-09-06 by lindoran