fbergama/pigfx #39: Sixel graphics possible?

Kirtai, 2020-09-05:
Any chance of adding [sixel](https://en.wikipedia.org/wiki/Sixel) graphics as used by the DEC vt330/340?

chregu82, 2020-09-05:
I had a look at sixel. What do you think would be the benefit? Do you have a specific need for this? As far as I saw, you could only draw a picture at the cursor position, nothing more. We already have more advanced graphic features.

lindoran, 2020-09-06:
I don't think there is a large library of vintage titles using Sixel encoding; and there's definitely nothing stopping somebody from wrapping those escape codes in a terminal program running on the host computer... The only thing i can think would be a utility add would be something along the lines of running a x11 session through frame serving running on a seperate server... and certainly there are cheaper and easier methods with a raspberry PI that is just running Linux... why run it on a potentially slower machine with all due respect it seems like needless feature creep to me.

Kirtai, 2020-09-07:
It was mostly for tools like [lsix](https://github.com/hackerb9/lsix) and the large array of tools based on [libsixel](https://github.com/saitoha/libsixel).

Kirtai, 2020-09-16:
Oh, there's also the simh emulator, which emulates old mini and mainframes like the PDP10 some of which used sixel. (Along with another graphics api called ReGIS)