fbergama/pigfx #41: small? Feature-Request: blinking Block-Form-Cursor

guidol70, 2020-09-13:
I successful use your PiGFX with Arduino Due running RunCPM - many Thanks.

So I would like to ask if it may be possible to get a flashing Block-Form Cursor?
It make me more easy to "find" the Cursor against the static white block ;)
Thanks again for reading/considering :)

chregu82, 2020-09-13:
Thank you for using PiGFX.
I think I'm going to do this. That's a nice little feature.

chregu82, 2020-09-16:
This is done in V1.8.1

guidol70, 2020-09-17:
Many thanks for the quick implementation ;)

I did wrote me the short CBLINK.BAS:
10 PRINT CHR$(27);"[?25b"
20 SYSTEM

and could start it with MBASIC CBLINK.BAS
(for MBASIC I had to set replaceLFwithCR = 1 in pigfx.txt)

Now Cursor is blinking fine 👍 
Is there a way to enable this on startup in pigfx.txt?

chregu82, 2020-09-17:
No, this can only be enabled with the escape code, like you did.

lindoran, 2020-09-18:
You can add  to autoexec.bat if you put the code into a .txt file and then use type filename.txt [nopage]

closed 2020-09-24 by chregu82

guidol70, 2020-09-30:
> You can add to autoexec.bat if you put the code into a .txt file and then use type filename.txt [nopage]

There is no real autoexec.bat in CP/M, but I did create a CBLINK.TXT
which can be used with the type command ;)
[CBLINK.TXT](https://github.com/fbergama/pigfx/files/5306825/CBLINK.TXT)

Normal filename on my system is CBLINK.ESC

lindoran, 2020-09-30:
Under my cpm 3.0 implemention on the Z80-mbc2 you can configure a autoexec
idk about other platforms, absolutely  it's not the .bat extension. I'll
have to take a look at the binary that loads on start up, maybe It is
possible to hack in the escape code into the version badge for cpm.
Otherwise if you compile your own cpm custom you can add it to any string
that echos out and it will blink the cursor.