fbergama/pigfx #42: How do you debug it?

wrljet, 2020-09-13:
How do you debug the code?

chregu82, 2020-09-13:
I always write perfectly working code, so there's no need for debugging.

----

Actually, no. I usually do this:
- Write code, if it compiles, I test it.
- If it fails, I think about it and review what I did.
- If it still fails, I write debug code which outputs some states and variables to the screen or to the serial port, if there's no framebuffer yet.
- I also recently added an exception handler which outputs the program counter and several registers at a crash. That's quite helpful sometimes.

wrljet, 2020-09-13:
I asked because I saw the JTAG directory with GDB stuff in it.

chregu82, 2020-09-13:
I never used JTAG, but someone did this in the early development.

fbergama, 2020-09-13:
Yea I did it with a cheap JTAG dongle at the beginning. Not particularly easy though. At this point it is probably easier to just print something at screen

closed 2020-09-16 by chregu82

wrljet, 2020-09-16:
Thanks, BTW.../Bill