fbergama/pigfx #44: color pallets? / typo in terminal_codes.txt

lindoran, 2020-09-29:
Is it remotely possible to create swap-able pallets?   Looking at porting old code from mode 13h, and I noticed the 256 color Xterm pallet is close but the order is different due to the way that it is organized; so old code which uses a decimal color value (like the PiGFX interface) will output different colors.  This facilitates the need for at the very least a conversion table (256 bytes of information).  But even then the colors are off slightly.  For example mode 13h (VGA / MCGA) color 1 is a blue hue RBG 0,0,170. The closest equivalent in XTerm is 19, which is 0,0,175.  I guess what I'm asking is could a set of control codes be implemented to replace a color something along the lines of esc[68;5;<color to replace>;<red value>;<green value>;<blue value>m.  This would be reset to defaults by esc [m.  

Alternatively a few select-able default pallets would be nice, like 13h (VGA/MCGA), CGA, EGA and TANDY perhaps? I can help with researching what the default values are, i have collected some links with good information as to what the RBG values are for each.

also I was looking over /doc/terminal_codes.txt there is a typo the clear sprite code which reads esc[#<idx>;x, but should read esc[#<idx>x.

Its probably a big ask, I was just wondering

chregu82, 2020-09-30:
I think this would be a very nice feature to be able to swap the color palette. I would probably define different palettes like Xterm, VGA, maybe others too. Xterm would be the default. I also could imagine to define a custom palette in a file on the compact flash. Palettes could be switched with a control code.
Could you provide the links you wrote about?

lindoran, 2020-09-30:
absolutely -- here is a link with the RBG values to decimal color number
for VGA (mode 13h): https://www.freebasic.net/forum/viewtopic.php?t=17474
about mid way down there is a full broken out chart in ascii for the 256
color palette in order with RGB values.-- I wish there was a better example
of this, perhaps I will have to make one ... I really looked for a long
time. This is the best I could find. This is the less than helpful
wikipedia article but it does have good other related information
https://en.wikipedia.org/wiki/Mode_13h.  This is yet another page on the
VGA mode 13h palette:
https://www.fountainware.com/EXPL/vga_color_palettes.htm

Here is ega, with the RGB values in hex
http://www.shikadi.net/moddingwiki/EGA_Palette.  To change from RGB take
the hex value, each 2 digit pair is the color value in decimal.  so 1 -
blue #0000AA is 0,0,170.  I think this pallet is very close to the first 16
of mode 13h (vga or MCGA).

here is a link to the C64 color pallet and the color values by decimal
color numbers (a chart near the bottom) -
https://www.c64-wiki.com/wiki/Color

This is for CGA: https://en.wikipedia.org/wiki/Color_Graphics_Adapter --
the values are listed in HEX for the character mode colors, but they are
also the same colors for the composite mode for CGA .... There's also a
good breakdown of the various 4 color pallets for graphics mode.  this
might be useful to have as well.

this is for the Tandy Graphics Adapter found on the tandy 1000 / PCjr:
https://en.wikipedia.org/wiki/Tandy_Graphics_Adapter

it's kind of a rabbit hole lol; I hope this helps :)

chregu82, 2020-09-30:
Please take a look at the develop branch. I implemented the ability to change the palette to VGA or C64. CGA and EGA colors seem to be part of the standard VGA color table.
I am unable to test this at the moment, as I don't have a Pi with me. So maybe it doesn't work at all.
Feel free to test it.
At the moment it's not yet possible to load a custom color palette.

lindoran, 2020-09-30:
I'm at work for 10 hours I'll look whebt I get home thanks!

lindoran, 2020-10-01:
the VGA pallet works as does the C64 Palette testing was done on the Pi-0 1.3 I ran a fractal from source and compared to output from the authors web page.  all appears to be working and well.  I wish I could test on a larger pi but I don't have one with functioning TX pins at the moment.  
(C64)
[image]
(VGA / mode 13h)
[image]

lindoran, 2020-10-01:
as stated, I didn't try custom but everything else seams to work so thanks :)

closed 2020-10-01 by lindoran

chregu82, 2020-10-03:
Custom palette is done now too.

lindoran, 2020-10-03:
I'll check it out in a bit :)

lindoran, 2020-10-03:
Custom is working on the pi-zero, manually programed in a mock 4 color CGA mode it's working perfectly,  :) thanks for this! really pushes this project into a new light, very few terminal's like this.  👍