fbergama/pigfx #46: mode4h / 320 x 200 -- apears off screen

lindoran, 2020-10-03:
was testing the CGA 4 color resolution with a custom pallet I don't know if this is a limit of the PI, the monitor or the software ; but when I do 320 x 200 some of the pixels draw off the screen; this issue is not present in 640x480 or 320x240.   this issue happens even with the Xterm pallet. this is with composite output.

lindoran, 2020-10-03:
see how the image is off center:
[image]

320x240 it's centered in a 4/3 aspect on the screen:
[image]

lindoran, 2020-10-04:
Looks like the same issue with 640x350 (only more of it renders off screen and is far shorter top to bottom). The upside is 640x480 / 320x240 are likely to be the most used resolutions due to this being evenly scalable to other common resolutions.

chregu82, 2020-10-04:
Actually I don't think there's much I could do about this. I tried different resolutions on a Pi 4 with a VGA monitor connected to the HDMI port. What this does if I change to a resolution which is not 4:3 is, it adds top and bottom black bars to get it to 4:3 again.
This is all Pi firmware related. I only can choose a resolution for the framebuffer.
I don't have a composite monitor to test with. It's probably even different with a 16:9 or 16:10 monitor connected with HDMI.

lindoran, 2020-10-04:
I figured something like that was the case. Those other resolutions are
less useful so I don't think it's a huge deal, but if it's a known issue my
only thought would be to update the document or disable those resolutions
as they don't perform as expected on other monitors. Thanks for the answer
🙂

willtco, 2021-09-16:
That is most probably your monitor (it certainly looks a few years old), it should have size and position controls which you can use to correct it.