fbergama/pigfx #47: USB Keyboard Support

Gonzo-XIII, 2020-11-15:
Not sure if I'm doing something wrong or if I haven't loaded some driver or such...
When I have no keyboard attached to the Pi, it seems to go through all the motions very quickly with a nice startup screen, like this:
[image]
But if I do actually have any USB keyboard (I've tried a few, including wireless ones) it freezes at the "Initialising USB" message and doesn't come back.
I read in the Readme.MD description that the USB support is provided with the USPi library. Is that built in, or is it another file that needs to be included on the SD card, or...?

chregu82, 2020-11-15:
USPI is builtin, so there's no need for any additional files.
You are using a very old version of PiGFX, I suggest you try the latest release. There are lots of bugs in the old version. I'm quite sure that this somehow crashes on USB initialization.
What Pi version are you using?

Gonzo-XIII, 2020-11-15:
Hmmm... I downloaded the files today from this repo... I might have picked the wrong file or something... I'll try again and report back.
I'm using a Pi Zero but really wouldn't know for certain how to tell what version it is. From memory its a 1.2 or 1.3 maybe? It's not the version that has wireless.

chregu82, 2020-11-15:
This is the latest version:
https://github.com/fbergama/pigfx/archive/V1.8.2.zip

Gonzo-XIII, 2020-11-15:
OK, I found the cause of the confusion.  The documentation about the Pi Zero card from Tindie sends you here:
https://rc2014.co.uk/modules/pi-zero-serial-terminal/
That page has instructions on setting up PiGFX, with a link to here:
https://github.com/fbergama/pigfx/tree/afcfc39647f1368b374883e120fe9da622b33a32
Which to a newbie, for all intents and purposes looks like the legit PiGFX repo. It isn't. Not sure what it actually is, but it has all the old versions of the software.

After tripping over (no idea what I clicked but I ended up in) the proper root folder for this repo, it looked different from what the first page looked like and I saw it had more checkmarks on the Todo list and... I said "haaaang oooon"... Then I read your message above about being an old version. You were correct, good sir. 👍 

All working great, after I had to edit the config file to convert LF or CR. (I reckon that should be the default.)

Last question: I tested it with a wireless keyboard... The kind that has a RF dongle and should by rights be a direct connection without any drivers... but it didn't work. Anyone had any luck with one of them or simply not possible to make them work?

closed 2020-11-17 by Gonzo-XIII