fbergama/pigfx #48: Sample Z80 Code?

Gonzo-XIII, 2020-11-17:
I can't seem to find any sample Z80 code. Would be very helpful to get new users started. 

Don't need anything complex... Just to show how escape codes are sent out to set some colours or the like.

chregu82, 2020-11-17:
You might have a look at this: [https://github.com/RC2014Z80/RC2014/tree/master/ROMs/pigfx](https://github.com/RC2014Z80/RC2014/tree/master/ROMs/pigfx)

Gonzo-XIII, 2020-11-18:
Perfect! Thanks...  I think that is worthy of having it linked to from this Repo!

ProgrammingCube, 2021-04-26:
I'm working on an Aztec C library for the escape codes, I need to comb through the code because some of the commands don't match what happens on screen (I suspect that a couple commands got changed without the documentation)