fbergama/pigfx #54: Inconsistent assembly code  suffixes

rkincaid, 2021-04-12:
I've been trying to build PiGFX on MacOS and it kept failing on asm.S. After a bunch of head scratching I figured out the make target is specified as $(SRC_DIR)/%.s and the gnu tools on MacOS are case dependent. So it had no target for .S hence the make failure. When I changed .s to .S, asm.S compiled correctly, but then I realized there are other assembly files with lowercase .s!  Either the makefile needs to reference upper and lowercase suffixes or the source files need to be consistent with either .s or .S and a matching target in the Makefile. I would think this would be an issue on Linux as well, but probably not on the PC which you are apparently using most of the time.

chregu82, 2021-04-12:
Damn, you're right. I'm using Windows at the moment. Just tried compiling it on a Linux VM and failed too. I'll take a look at it.

chregu82, 2021-04-12:
Please check out the develop branch and try if it's fixed.

rkincaid, 2021-04-13:
Build works now with a fresh copy of the develop branch on MacOS. You made the same change I did to the Makefile. Thanks for addressing this so quickly.

closed 2021-04-13 by chregu82