fbergama/pigfx #56: replaceLFwithCR + sendCRLF don't do anything

ProgrammingCube, 2021-04-22:
My issue right now with pigfx is the fact that for whatever reason, pressing Enter on the plugged-in keyboard sends a cryptic sequence:

<unknown>
LF

So, I thought, maybe its just some strange issue that can be forced as fixed if I change one of the options in the txt file.
However, neither options have  any effect on the Enter key.

chregu82, 2021-04-22:
I'm really sorry that I cannot reproduce this issue. Has anyone else had that problem?
Did you ever try connecting the Pi to a Terminal app like Putty and see what it sends?

ProgrammingCube, 2021-04-22:
okay, I have it probed. It just took me a bit to figure out how to probe it.

This is what I did:
[putty.log](https://github.com/fbergama/pigfx/files/6361447/putty.log)

Connect the uart1 port to Putty, and set it up to log everything (I take it that means that things sent over will also be sent back to the terminal, so you'll see double)

NOTE: I'm using a z180, so it has a built-in uart port, thats why I'm logging from the z180 card, not from the pigfx card.

Then, I hardware rebooted the z180 (you'll see romwbw)
Then, I booted into zpm (It works nicer with the backspace key, also its habit)

Then, I changed to A: drive, and hit the ENTER key on the plugged in keyboard 4 times.
Then, I changed to the B: drive and hit Ctrl-M 4 times, the first which was a confirmation of the B: command.

Here's the putty.log:

chregu82, 2021-04-22:
I always assumed that you connected a keyboard to the Pi and were talking about, that the keyboard input from the Pi to the SBC was not working correctly when pressing return. Now I' confused. What's your setup? Are you talking about data from the Pi to the SBC or from the SBC to the Pi?

ProgrammingCube, 2021-04-22:
Basically, I just had to choose a different probe point because the z180 has lines directly to the rc2014 bus. This is how the data gets to the z180:

`USB KEYBOARD
 |||
 vvv
PIGFX   -->> HDMI MONITOR
 |||
 vvv
RC2014 BUS BOARD
 |||
 vvv
Z180 Board  -->> USART1 PROBE (Running Putty, I'm not interacting with it but just pulling data)`

chregu82, 2021-04-22:
These parameters change the output of the TX pin on the Pi. You seem to look at the output of the RC2014 to the Terminal. Please monitor the TX pin of the Pi.

ProgrammingCube, 2021-04-22:
Ah, do you mean have the RX pin on the pi connected with the jumper, but set the tx jumper to the uart port on the pigfx card?

chregu82, 2021-04-23:
No, there's no need to alter any jumpers. Just monitor the serial line FROM the Pi to the RC2014. You always sent me the data which gets sent from the RC2014 TO the Pi.
There are 2 datalines!
One is for the keyboard input and one is for displaying.
It's abolutely possible that you don't see any difference after changing the settings on the Pi, because you are looking at data that is sent from the RC2014 and NOT from the Pi.

ProgrammingCube, 2021-04-23:
I tacked on 3 wires to the back of the pi, I'll do a putty pull and send you that.

ProgrammingCube, 2021-04-23:
Okay, I got the log.
You're right, it does look a bit different. This is the log of the TX pin from the back of the Pi.

So what I did was this:
1. Hook up the TX and Ground to putty.
2. Rebooted the machine
3. Hit Z to boot to zpm
4. Hit the Enter key 4 times
5. Typed out C: to change to the C: drive
6. I confirmed C: by pressing Ctlr-M this time.
7. I typed out snake
8. I confirmed snake.com by hitting the Enter key this time (note this is sending a command)
9. I'm in Snake now. In order to start, it says Press Enter.
10. I press Enter 5 times
11. I type out a 'b' as a spacer
12. I start snake by pressing Ctrl-M
[putty.log](https://github.com/fbergama/pigfx/files/6368442/putty.log)

chregu82, 2021-04-24:
According to the Log, Enter always genereates 0A, which is LineFeed. That's what is should do if both parameters are set to 0 (sendCRLF and replaceLFwithCR).
Ctrl+m generates a 0D, which is CarriageReturn.

ProgrammingCube, 2021-04-24:
I'll change replacelfwithcr to 1 and send you the log

ProgrammingCube, 2021-04-24:
Okay, I checked and double checked that this time, I set replacelfwithcr to 1.

Here's the log.
The steps are the same as above, but I accidentally misspelled snake so there's an m and a backspace there.
[putty.log](https://github.com/fbergama/pigfx/files/6369679/putty.log)

chregu82, 2021-04-24:
Ok it looks like it acts the same. So enter still generates LF.
What Pi model do you use? What version of PiGfx are you running?

ProgrammingCube, 2021-04-24:
I'm using a Pi Zero. I'm using a carrier board for the rc2014
[https://www.tindie.com/products/semachthemonkey/raspberry-pi-zero-serial-terminal-for-rc2014/](url)

I'm using the most recent(?) version of pigfx. Its in the bin folder of the master branch (main, now.......)
I've attached whats on my SD card, from the last test. Nothing is changed on it.
[bootcode.zip](https://github.com/fbergama/pigfx/files/6370216/bootcode.zip)

chregu82, 2021-04-24:
This drives me crazy. I tested it once again on a Raspberry Pi Zero V1.2 with the files from your zip. It sends 0D on enter, like it should if replacelfwithcr is 1. I really don't know what else to try.

ProgrammingCube, 2021-04-24:
Whats your baud rate? Mine is 115200, because thats what romwbw requires

chregu82, 2021-04-24:
I used exactly your config file, so I tried 115200 too. I just tried it with a Pi Zero W. Same thing, it works as it should.
Does it even load your config file? Could you post a screenshot of the PiGfx boot screen?
Are you able to change other things in the config file, e.g. the baudrate or the keyboard layout?

ProgrammingCube, 2021-04-24:
Okay, we're narrowing it down.

There's a weird bit about not reading a valid filesystem, and changing to uk doesn't work.

I tried formatting as both Fat32 and FAT, is this right?

I'll poke around for another card and try that.
[image]

ProgrammingCube, 2021-04-25:
YOOOOO

So I found a new sd card, formatted it as fat32, AND IT WORKS

Guess my card was just busted.
Now I can focus on building a standalone system!

chregu82, 2021-04-25:
Wow, that was a hard one.
I see that there are 2 partitions on your SD card. You could try to delete one and see if this fixes it.

ProgrammingCube, 2021-04-25:
Thanks, a total format/partition manager fix did it!

There's still a few other quirks I'd like to work out with pigfx, but I'll see if I can't work them out on my own first.

closed 2021-04-25 by chregu82