fbergama/pigfx #57: Compiling Error

ProgrammingCube, 2021-04-26:
I think I have the arm cross compiler toolchain set up correctly, but I always seem to be "that guy" when setting up dev environments...

I got the uspi cloned into the right directory and there's no error from gcc, but I am getting these errors:

`/bin/sh: 1: [: 4: unexpected error`
[image]

chregu82, 2021-04-26:
It looks to me as if this part of the makefile doesn't work:
kernel: pigfx.img
	@if [ $(RPI) == 1 ]; then \
		cp pigfx.img bin/kernel.img; \
	elif [ $(RPI) == 2 ]; then \
		cp pigfx.img bin/kernel7.img; \
	elif [ $(RPI) == 3 ]; then \
		cp pigfx.img bin/kernel8-32.img; \
	else \
		cp pigfx.img bin/recovery7l.img; \
	fi;

But I have no idea why...

chregu82, 2021-04-26:
Compiling on Windows is completely different than on linux. The makefile isn't used at all on Windows. There's a batch file that handles compiling.

ProgrammingCube, 2021-04-27:
Interesting, my friend was able to compile on linux no problem. I tried downloading and compiling on my home pc, but the same error popped up. Is it possible its a toolchain issue?

chregu82, 2021-05-04:
I'm going to close this if you don't mind.

closed 2021-05-04 by chregu82