fbergama/pigfx #60: [Feature request?] Change baudrate interactively

IrregularShed, 2021-05-28:
Hi

I love this project - it's given purpose to a couple of original Model Bs and some _very_ small SD cards (never thought a 128MB SD card would be useful again!).

A question though: would it be possible to have a way to change the baudrate at boot time (or any time, frankly) with a key combination, or maybe pulling a GPIO pin to ground, with the value in the config file the default? I have a lot of devices that require 9600, a few less that require 115200, a couple at 19200 and one at 38400. Being able to change the rate as required without having to edit the config on a PC would have an enormous benefit - right now I've got two SD cards to swap with the only difference being the baudrate value, which works but is another thing to lose or damage.

Steve

chregu82, 2021-05-28:
Hi Steve
That's a great idea. I just did it. You can give it a try by using the binaries in the filetransfer branch. The baudrate gets changed by pressing the print screen key.
Greetings Christian

IrregularShed, 2021-05-28:
Fantastic - I'll try it out as soon as I can!

IrregularShed, 2021-06-01:
That works beautifully - thanks for the change! 😁

chregu82, 2021-06-01:
Nice, you're welcome.

closed 2021-06-01 by chregu82