fbergama/pigfx #63: Bypass PIGFX card without removing it?

ProgrammingCube, 2021-07-15:
I was wondering if there was an idea for a hack where I can hit a switch or button and have the uart not be intercepted by the pigfx card, but instead pass out through the normal UART port (Z180, it has its own UART breakout on the CPU board and I use that to transfer files easily to/from Windows).

Would just a switch on the UART jumpers be fine for this card:
[https://www.tindie.com/products/semachthemonkey/raspberry-pi-zero-serial-terminal-for-rc2014/](url)

chregu82, 2021-07-15:
I don't see a reason why a switch would not work.

ProgrammingCube, 2021-07-18:
So, when I have the system switched to the ftdi mode, I can get serial out, but it seems keypresses don't register.