fbergama/pigfx #64: Compile fail.

willtco, 2021-08-03:
I downloaded the zip file 2.7mb then used ./makeall which immediately fails as the uspi directory is empty,  comparing to my original download (2019  8.1 mb)  which had significant subdirectories in uspi.  The original software would hang if left for several hours hence my wish to upgrade,  is there a full package available to download?

chregu82, 2021-08-03:
You could use either the binaries from the releas page or if you want to compile it yourself you would have to download the uspi submodule and put it into the uspi directory.
Github does not include submodules in the zip.

willtco, 2021-08-03:
Ok sorted, the original 2019 load included uspi, this one does not,  since I greatly modified several files I needed to re-compile. All sorted now  thanks.

closed 2021-08-03 by willtco