fbergama/pigfx #65: unable to correctly set baudrate.

willtco, 2021-08-29:
I have followed the instructions to the letter,  yet when booting the pi it indicates that the uart is expecting data at 115200 rather than the 9600buad set in pigfx.txt.  I am using a pi-zero.

chregu82, 2021-08-29:
Please read this issue: https://github.com/fbergama/pigfx/issues/56

Does it even read the filesystem on the SD card?

willtco, 2021-08-29:
Thanks for the reply, The pi boots up fine, at the bottom of the screen there's a message indicating the uart is waiting for 115200 8 n 1  - when I press a key on the keyboard a string of garbage appears, it seems clear that although 9600baud is set in the .txt file it only wants to play the 115200 baud rate.  The issue you mention seems to be something quite different.

chregu82, 2021-08-29:
Please post a picture of your boot screen. I'm quite sure that PiGFX is unable to read the SD card and therefore cannot read the config file and therefore uses the default baudrate of 115200.

willtco, 2021-08-30:
/media/dadmin/NEW VOLUME/DCIM/100CASIO/CIMG5316.JPG

willtco, 2021-08-30:
[image]
This photo is of a compile of the downloaded uspi & piGFX zip files without any modification.  afterlooking through the source I found the default is located in 'config.c'  - when changing the default  .uartBaudrate to 38400 my z80 card was able to send data to the pi and it was correctly represented on screen. However  the usb keyboard was inoperative (except for the function keys which produced a few characters of garbage. My 2019 pigfix has many updates which I hope to transfer to the latest edition hence my need to compile from source - I thank you for your patience. 

> [[image]](/willtco)

chregu82, 2021-08-30:
It's like I thought. PiGFX is unable to read the SD card. You seem to have 4 partitions on your card. This doesn't work. Please delete all partitions, add a single new one, format it and put PiGFX on it. It should work afterwards.

willtco, 2021-08-30:
[image]
Yes I thought that but making a second card produced the same results, Using linux mint to examine the card shows only a single FAT partition,  as below, the card was formatted under windows, it is a 2GB card as that's the smallest one I have. It contains only the few files for piGFX.

chregu82, 2021-08-30:
Could you use the windows tool for checking the partitions on the SD card? There seem to be more than your linux mint shows. If there's only one FAT formatted partition it should work fine.

willtco, 2021-08-30:
[image]

WIndows shows only a single partition as well - very confusing.

chregu82, 2021-08-30:
Does this SD card produce the boot screen you showed earlier?

willtco, 2021-08-30:
I think (dare I say it) that the problem may be with your MBR reader,  for large partitions it shows  "MBR: found total of 0 partition(s)"  - and for a small partition it seems to be reading the wrong sector and mis-interpreting ascii messages as partition records. - if you can point me to the source code for this part I'll take a close look. - I knew 17 years in the disk industry might come in handy one day!

willtco, 2021-08-30:
yes same one.

willtco, 2021-08-30:
Can you upload the MBR of a working card?  - it should start with EB and end with 55 AA - I'll graft that onto a card, then format it - I think at some time the MBR - here is the MBR of the re-formatted 2gb card it clearly shows that Windows has replaced the true mbr  with the first sector of a partition.  If I can replace this with a good mbr from a 1GB card then reformat I believe that will cure the baudrate problem,  but whether it will also cure the USB KBD I'm not sure 
[image]

willtco, 2021-08-30:
Update - after grafting a 500mb Fat 16 MBR onto the SD card, & re-formatting - the first part of the problem is solved - 
[image]
Should I open the usb keyboard issue seperately?

chregu82, 2021-08-30:
I don't have a working MBR at the moment, that I could show you. Glad it's working now. Please open a new issue.

willtco, 2021-08-30:
Thanks for your help.

closed 2021-08-30 by willtco