fbergama/pigfx #66: USB keyboard feature not working.

willtco, 2021-09-03:
Downloading the latest PIGFX (as at Aug26th 2021) and USPI 2 - when compiled the pi is not responding to the keyboard inputs.  2 different pi(zero's) and 3 USB keyboards have been tried in various combinations with no effect.
[image]

chregu82, 2021-09-03:
- Have you tried the precompiled binaries?
- Have you used the linked USPI submodule (other versions may not work)?
- Have you looked at the TX line of the PI, if there is any activity if a key is pressed?

willtco, 2021-09-03:
Have now downloaded and tried the precompiled - same result no keyboard action.

chregu82, 2021-09-04:
I just tested it with a Pi Zero (without W) using this image https://github.com/fbergama/pigfx/blob/master/bin/kernel.img
I connected RX and TX on the Pi and was able to directly write to the screen with the USB keyboard. At the moment I have no idea why it would not work on your setup.

willtco, 2021-09-04:
Thats fair enough, all I can say is that it works fine with my 2019 (unaltered) kernel but not with the latest. I'll do a bit more work and report back if I find/cure the problem. Meantime if you can think of anything that has changed and may be significant between 2019 and today it might help.

willtco, 2021-09-04:
Ahh interesting, I see that you only get activity on the TX line if it's connected to the RX line,  whereas on the 2019 just using the keyboard gives activity on TX -  I suggest you might like to try hooking yours up to a pc maybe something like Teraterm?   If you have a scope look on your TX line when it' NOT connected to RX -  Or maybe find out where things have changed. Rgds.

willtco, 2021-09-04:
38.4K - here's a scope trace of what's on the TX line, looks like. It should be a square wave, but it's not being pulled low enough to register logic zero. It could be the TX pin is not properly configured tiup/dn or current capability. - As you can see the scope is set to DC with a x10 probe. Left hand marker is 0v.   I get identical results from a second Pi.
[image]

chregu82, 2021-09-04:
Ok I tried doing the same, still the mentioned (new) kernel image, 38400 baud, RX and TX connected together. TX is set to no pull up / no pull down, RX is set to pull up in uart.c.
To me this looks absolutely fine. What's your load on the TX line? Do you have a scope of the old software version?
It seems that you have some capacitive load on this line.
[image]
At least we now know that it's not a USB keyboard related issue.

willtco, 2021-09-04:
It may be a few days before I can respond fully, but thanks for the trace & info - your comment about capacitance made me think about the Scope probe I was using, - it's now in the bin.  with a second probe the waveform is fine.  I need to now take a closer look at the rest of my system. I guess at nearly 70 I should turn my hand to gardening!.

willtco, 2021-09-06:
OK 2019 load echoed keyboard characters to the screen. The 2021 version does not do this.  Thus the keyboard appeared  inoperative,  once I added some debug code to the target system it became obvious that characters were getting through but commands were not being processed,  this eventually was traced to the 2021 code sending an LF character whereas my target system would only action a command on receipt of a  'CR'  thus 2019 was fine, 2021 failed.

chregu82, 2021-09-06:
You can set the behaviour of CR or LF in pigfx.txt.

willtco, 2021-09-06:
Solved, - 3 problems, 1) a dud scope probe, 2) usb kbd appeared not working as there was no echo (2019 had echo?) - 3 I failed to spot CR had been replaced with an LF and I failed to notice it could be swapped in the config file. My thanks to Christian Lehner for pointing me in the right direction.

chregu82, 2021-09-06:
You're welcome.
Not sure what you mean with echo. PiGFX should display what gets sent to the RX pin. So it's your system which should echo the characters it receives from the USB keyboard.

closed 2021-09-06 by willtco