fbergama/pigfx #67: Pi/KBD freezes after several hours.

willtco, 2021-09-08:
I have noticed that if I leave my system running it freezes or locks up after  several hours,  no more keyboard characters are passed to the screen and even trying the caps lock fails to light the LED. I checked by connecting TX to RX by direct link at the PI, using the latest download and had the same results.

chregu82, 2021-09-11:
I don't seem to be able to reproduce this issue. My system is now running for 10 hours and still no freezing.

chregu82, 2021-09-11:
Nice, after 13 hours runtime my keyboard also freezes. The Pi is obviously still running as the LED still blinks. How on earth should I find out what causes this?

willtco, 2021-09-11:
OK  I found that my Z80 could still send data to the pi and have it put on screen, I found that caps lock would not work.  I understand the USB 5V can be switched on off by the Pi, this might be the first thing to check, after that is there a watchdog time out?  - a buffer full?   and after that who knows?  - but glad I'm not hallucinating on this one! :)

willtco, 2021-09-12:
Over the time I have seen this happen the time taken to 'freeze' has been as little as 2 hours or as long as 15 hours, and all on the same set-up.  I am wondering whether there could be an outside influence causing this such as some sort of power disturbance?  - since the power supplies for PI are not of the highest quality  I may try a bench PSU  and a 12 volt monitor.  I would expect the system to freeze after an exact time if repeat tests are identical.

willtco, 2021-09-15:
Another clue,  I have been running with a PS/2 keyboard which has also frozen after 21 hours of just sitting there.

willtco, 2021-09-15:
Running on a PS/2 keyboard for 21 hours (with no USB KBD connected) has also frozen.  And again pressing caps lock did not illuminate the local LED.

chregu82, 2021-09-15:
That's quite interesting. Then it's not related to USB or the uspi submodule. I'll probably should add some debug code and see where keyboard input gets lost.
LED handling of the keyboard happens in software as reaction to the pressed key.

willtco, 2021-09-15:
I will do another couple of  runs  just to confirm  PS/2 is also affected, after that I'll tighten up the power supply and maybe add a sprinkle of capacitors.
[image]

willtco, 2021-09-16:
I've not yet been able to get the PS/2 to fail a second time (after 26hours) - I will be changing those wire links for a more permanent ribbon cable in case any of them caused a glitch.  So I'd hold fire on changing anything just yet for a few days till I can confirm whether PS/2 is actually affected.