fbergama/pigfx #69: Composite output

esalgado, 2021-11-24:
Hello!

is it possible to use the Composite output instead of the HDMI? I.E. to hook it up to an old TV

Thanks for this awesome project :)

chregu82, 2021-11-25:
Hi
As far as I know this should work out of the box.

esalgado, 2021-11-25:
Hi! Thanks for the reply, I have tried with a working Composite cable, tested with Raspbian. Does not work with PiGFX on Pi 4:

https://www.raspberrypi.com/documentation/computers/config_txt.html

> On the Raspberry Pi 4, composite output is disabled by default, due to the way the internal clocks are interrelated and allocated. Because composite video requires a very specific clock, setting that clock to the required speed on the Pi 4 means that other clocks connected to it are detrimentally affected, which slightly slows down the entire system. Since composite video is a less commonly used function, we decided to disable it by default to prevent this system slowdown.

> To enable composite output, use the enable_tvout=1 option. As described above, this will detrimentally affect performance to a small degree.

Thanks!

chregu82, 2021-11-30:
Have you tried putting a config.txt onto the SD card with enable_tvout=1 ?

closed 2021-11-30 by esalgado