fbergama/pigfx #70: Something not right, with apps that request number input and do not recognize enter key.

kwpodolsky, 2021-12-08:
System: SC126 - WBW Rom 3.0.1 (alto same issue with 2.9.1)
Rom running Z-System or Basic
PiGFX version (latest bins released 8 months ago).

I'm using a standard USB keyboard into the Raspberry Pi Zero's usb port.
98% of things seem to work fine with this setup except for when I try to run TurboPascal Tinst or MBasic.
Tinst prompts for [S]creen installation->Which terminal(enter no. or ^Q to exit) <valid no is 1-33>
I enter a number and hit enter.... nothing happens.  I can BS over the entry and hit other numbers... I can even do CtrlQ to exit.
But entering a number and hitting the <Enter> key... is not recognized.

If I run Nascom Basic, I am prompted for ...   Memory top?
I can enter a number, but AGAIN, the <Enter> key is not recognized or just ignored.

I NEVER saw this issue when I used a real usb interface to directly communicate with the SC126 board.

BTW,  I have also tried to send Ctrl-L and Ctrl-J instead of using the Enter key.   No help.

Is there some configuration parameter that needs to be set?

Thanks in advance for any advice.

chregu82, 2021-12-09:
Hi
Usually the problem is that a piece of software expects either a CR or a LF character as enter key. It is possible to configure the character which is sent by PiGFX within pigfx.txt. Please look for "replaceLFwithCR ".

kwpodolsky, 2021-12-09:
Yeah, I thought that might be it, but the same programs work fine when I connect via USB->Serial and run TeraTerm.
Settings on TeraTerm are, send CR only, and yet these apps work fine.
And, as I said, I tried doing Ctrl-L(0x0d) and Ctrl-J(0x0a), and just Ctrl-J, on the keyboard and they didn't work either.

I was thinking that it might be the way these apps try to get their input, but that makes no sense why they will take the numbers or even the Ctrl-Q code, but not process the enter key.

Believe me, I would not have posted here, if I hadn't exhausted all the debugging thoughts possible (I've been programming since 1976.  From IBM mainframes to microcontrollers).

I could dig into the code to see what get's sent (or jammed into RX pin) when Enter is hit, but I'm not sure that will prove useful to me.

I will try the "replaceLFwithCR" but I don't believe that will make a difference.
If it does/n't   I will let you know.
Anyway, if you have any other thoughts, I'll try them.

kwpodolsky, 2021-12-09:
Ok... makes NO sense to me, but I changed the config file to "replaceLFwithCR" and now things seem to work on the two apps that were failing.

Thanks for your help.

Kerry

closed 2021-12-10 by chregu82