fbergama/pigfx #71: New Line/Enter not working in certian situations.

smarinel, 2022-02-17:
I have the newest build running on a Pi Zero W/RC2014 using CP/M

I can generally type fine at the CP/M prompt but inside ZDE16 or Zork1 I cannot hit enter to submit a command or go to a new line. I tried messing with the LF/CR options in the config file but nothing was working. Any idea what could be going on here?

chregu82, 2022-02-18:
Every single time I heard about such an issue, it was LF instead of CR or CR instead of LF. Are you sure that your config file is read during boot? Please check your PiGFX boot screen.

chregu82, 2022-02-18:
It seems that this is the problem. It fails reading the filesystem and because of that it can't read the config file. You should make sure that there is only one partition on the card or try using a different card.

closed 2022-02-18 by smarinel

paulwratt, 2022-02-20:
@chregu82 

What was the sd-card size that failed, and the one that worked?

Also what was the filesystem that failed and worked (it might be related to the size)?