fbergama/pigfx #73: Pi pico ?

jluke1, 2023-01-05:
Would it be possible to run pigfx on a pi pico ?
Is there some way to generate a binary that could be
down loaded to the pico flash ? Does the pico have the
resources to run pigfx ? Less memory, slower clock.

Thanks for a very nice piece of software.
Regards
John Luke

chregu82, 2023-01-05:
Hi John
Unfortunately it's not possible to get PiGFX running on a Pi Pico. It's a completely different platform, it's missing an SD card port and doesn't even have HDMI output.
Regards
Christian

tawez, 2023-01-05:
I would agree only with that it is different platform (ARM Cortex M0+).
There is SPI on board, so SD card can be attached. VGA or HDMI can be added as well. Pimoroni Pico DV Demo Base is a good example (there is also VGA version).
The question is if the same code base can be adapted, or should it be completely new project.

jluke1, 2023-01-05:
I was afraid so had forgotten about HDMI. A pity though.
Looks like a pi zero would be the best low end alternative.
Should this issue be closed ?

Regards
John Luke

RFC2795, 2023-01-05:
The worldwide Pi Shortage recently meant that I wasn't able to supply the Pi Zero Serial Terminal for the RC2014 (The device PiGFX was originally written for), so I turned to the Pi Pico, and now have the Pi Pico VGA Terminal that runs PicoTerm https://z80kits.com/shop/rc2014-pi-pico-vga-terminal/

PicoTerm runs very similarly to PiGFX, although due to a lack of RAM, it can either run 80 column mono or 40 column colour.  It has VGA output, although there should be an HDMI compatible version out soon.

chregu82, 2023-01-06:
I think this would be the best way to go if you want to use a Pi Pico.

closed 2023-01-06 by chregu82