fbergama/pigfx #75: Qemu instructions and scripts out of date

F1r3Hydr4nt, 2023-08-07:
The original qemu dependency was merged: https://github.com/Torlus/qemu/tree/rpi And now the scripts are out of date it seems, I cannot connect to the vnc :/

chregu82, 2023-08-07:
I'm afraid, since I worked on pigfx I never used qemu. So I don't know how it works or what it needs to get it working again.

bkg2018, 2023-08-08:
Hello, back in 2020 when I worked on pigfx too I was unable to use Qemu as told in instructions. I guess instructions are long obsolete.

To me it's hard to imagine any reliable debug method other than actual run on a Pi for real hardware issues like interrupts or boot. But the purely software functionalities like ANSI console functions would benefit from an emulator.

----- Mail original -----